makerbase-mks/MKS-PI #3: nmtui doesn't work

acguardia, 2022-07-13:
mks@mkspi:~$ sudo armibian_config
sudo: armibian_config: command not found
mks@mkspi:~$ nmtui
-bash: nmtui: command not found
mks@mkspi:~$ sudo su
root@mkspi:/home/mks# cd
root@mkspi:~# nmtui
bash: nmtui: command not found
root@mkspi:~#

makerbase-mks, 2022-07-19:
@acguardia, yes, we have uninstalled the nmtui, as it seems it has conflict with the dhcpcd. 
Without nmtui, you can also use "armbian-config->Network" to config the network.
If you still want to use nmtui, you can install it, but had better to uninstall our service:
`systemctl disable makerbase-net-mods.service`
`systemctl disable mks_wifi@wlan0`
`apt install network-manager `

I don't find the problem of "apt upgrade", maybe you can check the network?

jakep82, 2022-07-22:
Same issue here.  I'm unable to resolve any web address making the board completely unusable.  It's also not possible to install network-manager without working host resolution, and I don't know how to solve this using armbian-config.

makerbase-mks, 2022-07-25:
@acguardia @jakep82 ，we found it should be caused by the /etc/resolv.conf, you can try to modify the DNS on it:
`sudo sed -i "s/192.168.2.1/8.8.8.8/g" /etc/resolv.conf`

makerbase-mks, 2022-07-26:
The new image has been updated, and fix this issue:
https://drive.google.com/file/d/1-L2HId9kAp05xsv309gqI9QjSpi3GUCU/view?usp=sharing

closed 2022-07-27 by makerbase-mks