makerbase-mks/MKS-PI #4: Sources/linuz kernel?

guysoft, 2022-07-19:
Hey, 
Creator of CustomPiOS here.
Was wondering if you can provide the omage and linux kernel sources?
Is the kernel mainline?

Thanks!

makerbase-mks, 2022-07-25:
@guysoft ，we use the [Armbian mainline](https://github.com/armbian/build).

guysoft, 2022-07-25:
Thanks for clearing that out, might be a good idea to note that in the README :)

closed 2022-07-25 by guysoft

redrathnure, 2022-12-25:
@guysoft , I am trying to build Armbian based on current (5.15) and edge (6.1) kernels. Please see [redrathnure/armbian-mkspi : custom/mkspi_23.02.0-trunk.0073](https://github.com/redrathnure/armbian-mkspi/tree/custom/mkspi_23.02.0-trunk.0073) for more details. 

If you are interested, testing, review and adjustments are welcome :)

guysoft, 2022-12-25:
@redrathnure Thanks for the heads up. Don't think I have that time for that, not the hardware.