makerbase-mks/MKS-PI #10: problem connecting with hostname

LYRA3D, 2022-08-29:
mkspi.local is not working.
In the same computer, raspberry model 3 works fine with fluiddpi.local

Tested on 2x computer and iphone with same results

TheFeralEngineer, 2022-11-25:
I usually just connect with mkspi and the port. I'm trying to figure out why I can't change the hostname...

formy2000, 2023-01-08:
> I usually just connect with mkspi and the port. I'm trying to figure out why I can't change the hostname...

I was able to change my hostname via the sudo armbian-config command. Although via http it is still failing to resolve the hostname.

TheFeralEngineer, 2023-01-08:
> > I usually just connect with mkspi and the port. I'm trying to figure out why I can't change the hostname...
> 
> I was able to change my hostname via the sudo armbian-config command. Although via http it is still failing to resolve the hostname.

yeah, I used sudo hostnamectl set-hostname to get the hostnames on my boards to change the hostname and was able to connect to them for weeks, did a fresh install on both boards and now neither will resolve with hostname and only one displays the new hostname on my router list, the other still says mkspi... and good luck getting an answer out of these guys...