makerbase-mks/MKS-PI #14: Spi Clock Increase breaks adxl345 support.

findus4fun, 2022-09-27:
From your Logifile:
Version: MKS-PI-Armbian_22.05.0-linux_5.16.20-20220726.img
MD5: 0DBFA0092315C3B8D5E1C99B573065F2
SHA-1: 9E6480EAEAB9AA6591C23DDB1B02EFA723AB45A9
CRC32: E6252C17
Modification: 
1. Increase the spi clock of PI-TS35 to _**25Mhz**_
2. Modify the default DNS server in /etc/resolv.conf to 8.8.8.8, so that there will be no networking failure when connecting to other hotspots
3. Fix the issue that "apt upgrade" will cause the image to be destroyed and klipper cannot be used

So Spi runs with 25 MHz. 
From ADXL345 Datasheet:

SPI
For SPI, either 3- or 4-wire configuration is possible, as shown
in the connection diagrams in Figure 34 and Figure 35. Clearing
the SPI bit (Bit D6) in the DATA_FORMAT register (Address 0x31)
selects 4-wire mode, whereas setting the SPI bit selects 3-wire
mode. The **maximum** SPI clock speed is _**5 MHz**_ with 100 pF
maximum loading, and the timing scheme follows clock polarity
(CPOL) = 1 and clock phase (CPHA) = 1. If power is applied
to the ADXL345 before the clock polarity and phase of the host
processor are configured, the CS pin should be brought high before
changing the clock polarity and phase.

Since PI-TS35 and adxl345 share spi bus (and spiclk) adxl345 does not work. I did test this in hardware.

How to fix this issue?

Is there a way to use a second spi for the adxl using some of the other I/O pins and the connectors?

Regrads
Jan

michalfrnoch, 2022-12-11:
i have same problem

formy2000, 2023-01-25:
Anyone find a fix yet? Similar to above comments, my screen goes white when the ADXL345 is connected

redrathnure, 2023-01-25:
Have you tried different image?

formy2000, 2023-01-25:
> Have you tried different image?

I am using the image that came with the MKS EMMC (Armbian 22.05.0-trunk). I am assuming they placed that image on their product. Or is that an incorrect assumption.

redrathnure, 2023-01-26:
> I am assuming they placed that image on their product. Or is that an incorrect assumption.

yes, they are. However I had several issue with it (non stable display, misconfigured network settings, random version of software and few more). Unfortunately MKS support have not help me (no answers at all) and I decide to build [a few custom images](https://github.com/redrathnure/armbian-mkspi/releases).

Last weekends I tested Jammy-current image together with MKS display, SKR 1.3 board (USB), ADXL345 sensor (SPI), WEB cam (USB) and few more devices - everything works fine. 

If you decide to try it, you will have to set up the system from the beginning and [install Klipper from scratch](https://github.com/th33xitus/kiauh). However, this will probably solve your original problem and get you more or less fresh (and standard) Armbian.

formy2000, 2023-01-26:
I will give your image a try this week. 

Also, I have 3 of the MKS Pi V1.1's. I found the blanks screen occurs when using the EMMC with the ADXL345 connected. On a different MKS Pi I had the same build but not through the EMMC and the screen works as it should when the ADXL345 is connected. 

I am going to do more tests this evening.

Thanks for looking into this. It is unfortunate that MKS support is not involved more.

findus4fun, 2023-01-26:
Hi,
my MKS PI is also runnning from EMMC. Have not tried to run from SD yet. 

@redrathnure does the ADXL345 work in klipper using your custom image?

I got tired and just used my old bluepill setup because it has only one spi slave on the the spi bus. 

Thanks for providing custom images. I hope I will find time to test them soon, too.

redrathnure, 2023-01-26:
Yes, an ADXL345 via SPI connection works. The tricky part is not forgetting about `klipper-mcu` service. Please follow [Install the rc script](https://www.klipper3d.org/RPi_microcontroller.html#install-the-rc-script) and [Building the micro-controller code](https://www.klipper3d.org/RPi_microcontroller.html#building-the-micro-controller-code) instructions (or [`ADXL345/SPI Usage` chapter from the fork](https://github.com/redrathnure/armbian-mkspi/tree/custom/mkspi_23.02.0-trunk.0171#adxl345spi-usage))

findus4fun, 2023-01-27:
I just tried you latest image from sd card. So far it worked very well. Tomorrow I will hook up my adxl345 and test if that works.
Running from EMMC should be just flashing your Image to the EMMC. I will test that if my adxl345 is successfull.

Thank you very much for the effort you have put into building these images. Great Work.

formy2000, 2023-01-27:
> Yes, an ADXL345 via SPI connection works. The tricky part is not forgetting about `klipper-mcu` service. Please follow [Install the rc [...]

I tried your custom build on the EMMC. Unfortunately, the screen goes white when the ADXL45 is connected. I still get results with ACCELEROMETER_QUERY so I assume the ADXL345 will function.

As soon as I disconnect the ADXL345, the screen returns to normal. I might also add that I like this screen better than the one given with the MKS build provided on the EMMC.

My only guess is still that the ADXL345 makes the screen go white (blank) when the build is through EMMC. When the build is on an SD card the screen is fine when the ADXL345 is connected.

Considering the connection of the ADXL345 is temporary, I will do the accellerometer tests with the white screen and see if the ADXL345 is functioning and remove when not being used.

findus4fun, 2023-01-29:
I can confirm the screen going white when running from EMMC. 

Maybe the screen going white is hardware related, but that is just a guess.

I think the new image has a better screenhandling and so I will keep running the new custom image.

formy2000, 2023-01-29:
> I can confirm the screen going white when running from EMMC.
> 
> Maybe the screen going white is hardware related, but that is just a guess.
> 
> I think the new image has a better screenhandling and so I will keep running the new custom image.

I fully agree. The custom build is much better. Even though I had to add Klipper, Moonraker, etc from scratch, it is also my preference.
As for the white screen issue, I also agree that it is a hardware issue. Screen works fine when ADXL345 is connected and not using EMMC.
I doubt Makerbase will reply to my email.

findus4fun, 2023-01-29:
I just came across MPU6050 as a replacement for adxl345. 
This will be connected via i2c. 
I have one board I will try to connect it to the mks-pi i2c pins. 
Hopefully these will work toggether with the tft screen.

formy2000, 2023-01-30:
> I just came across MPU6050 as a replacement for adxl345. This will be connected via i2c. I have one board I will try to connect it to the mks-pi i2c pins. Hopefully these will work toggether with the tft screen.

Hopefully it works. Makerbase has just replied to me this morning and has acknowledged that it is an issue by stating the following:

**"This problem is because the transmission frequency of the adexl345 acceleration sensor is relatively high, which will interfere with the product.
The screen is transmitted through spi, so it may be affected, and the screen will be white. This may be a limitation of the spi transport. 
We're working on an HDMI screen that won't be affected by this."**

findus4fun, 2023-01-30:
I still doubt that they understand the problem. The TFT and adxl interface share the spi signals no matter if the image runs from SD card or Emmc.

redrathnure, 2023-01-30:
TFT, touch screen and SPI port (adxl345) share the same RX/TX lines. SD cart and EMMC are connected to different lines. I am not sure how adxl345 may affect EMMC. In theory, if they use software implementation of SPI and/or SD/EMMC interfaces, we may have lack or CPU/resources to handle all devices on the same time. But I just speculating here... However, you may check CPU and overall system utilization during working with ADXL345.

findus4fun, 2023-01-31:
> I just came across MPU6050 as a replacement for adxl345. This will be connected via i2c. I have one board I will try to connect it to the mks-pi i2c pins. Hopefully these will work toggether with the tft screen.

Hm, looking at the schematics you have either i2c are ethernet Leds. You have to solder resitors to use one or the other. I will stick to my setup with an extra blue-pill.