makerbase-mks/MKS-PI #16: Slow Wifi speed

ils15, 2022-10-03:
Comfast adapter has low wifi speed 
i need to investigate more, but appear a kernel related problem

RPI4 with 5GHZ 

[image]

MKS PI with 2.4 and comfast adapter

[image]

garbean7, 2022-12-01:
I'm having the same issue. Wifi with the comfast adapter is incredibly unstable and slow. Did you find a solution to this yet?

ils15, 2022-12-01:
No. And Markerbase didnt answer anything

garbean7, 2022-12-01:
Are you able to use the fluidd webpage via wifi? For me it looses connection to moonraker and sometimes the network every 30s despite a great signal. If connected to a ethernet cable everything works fine.

ils15, 2022-12-01:
i think yes
i reinstalled everything using kiauh, but speed still slow

garbean7, 2022-12-01:
My R8188EUS did not work properly or the wifi signal was too weak for it. It now works fine with another adapter.

ils15, 2022-12-02:
> My R8188EUS did not work properly or the wifi signal was too weak for it. It now works fine with another adapter.

which adapter?
have you tried any 5 GHZ adapter?

mrmainx, 2022-12-12:
I have similar problems with the comfast adapter. With LAN the network works as it should. But with the comfast adapter the network isn't really usable, as i get a lot of timeouts.

ils15, 2022-12-12:
> @ils15 The adapter I'm using now is called "AVM FRITZ!WLAN Stick AC 860". I'm running it with the driver "firmware-linux-free" and it is currently running well on 5 GHZ.

great, can you check model in terminal using

sudo lshw -C network

redrathnure, 2022-12-12:
Just FYI: I tried current image from Renegade board (Ubuntu Jammy + Kernel 5.15.82) together with Comfast dongle. 3 hours without any package drop and avg. up and down speeds are 10-12 Mbit/s . But MKS image has extremely unstable connection and 2-3 Mbit/s speed (same board, same dongle and same physical location). IMO looks like issue with provided image or kernel.

ils15, 2022-12-12:
> Just FYI: I tried current image from Renegade board (Ubuntu Jammy + Kernel 5.15.82) together with Comfast dongle. 3 hours without any package drop and avg. up and down speeds are 10-12 Mbit/s . But [...]

disable automatic set network channel in your router. i think it can be solve issue

i tried to compile image, but mks PR is so bad

redrathnure, 2022-12-12:
>disable automatic set network channel in your router. i think it can be solve issue

Already tried it. No luck.

> i tried to compile image, but mks PR is so bad

PR? Do you mean few latest  commits from https://github.com/makerbase-mks/armbian-build?

ils15, 2022-12-12:
> > disable automatic set network channel in your router. i think it can be solve issue
> 
> Already tried it. No luck.
> 
> > i tried to compile image, but mks PR is so bad
> 
> PR? Do you mean few latest commits from https://github.com/makerbase-mks/armbian-build?

yes

fbeltrao-br, 2022-12-12:
I identified at least three problems with this dongle.

1 - It do not fare well with wireless router auto channel selection (like channel=acs_survey in OpenWrt).

Solution: Fix your channels in your router settings.

2 - The dongle is not detected after a complete clean boot, needs to be removed and reinserted in USB port.

3 - The dongle suddenly looses RX/TX power.

Solutions: add these lines to /etc/rc.local and reboot.

usb_modeswitch -v 0x0bda -p 0x8179 --reset-usb
modprobe r8188eu rtw_power_mgnt=0

mrmainx, 2022-12-22:
I Found another problem with the dongle (rtl8188eu 0x0bda:0x8179).

On my fritzbox i had the 802.11b standard disabled. After enabling it, the dongle works without connection loss.

formy2000, 2023-01-06:
Reading this post kind of helped. I also had continuous moonraker drops and reconnects with the Comfast wifi dongle that came with my MKS Pi V1.1. I am in a location in my home where there is no RJ45 connections. I will try a different brand of USB wifi dongle tomorrow and hope that it works.....if not, sadly MKS will push me back to Raspberry

onolox, 2023-02-23:
I will try to mod my dongle to use a router antenna.

onolox, 2023-02-23:
> 
> Edit: I purchased a BrosTrend Linux USB WiFi Adapter dongle on Amazon and no more issues

how is the cpu usage? stays low?

formy2000, 2023-02-23:
> > Edit: I purchased a BrosTrend Linux USB WiFi Adapter dongle on Amazon and no more issues
> 
> how is the cpu usage? stays low?
> 
> I will try to mod my dongle to use a router antenna.

I have had no issues at all. CPU usage has been normal. I have two printers on separate MKS Pis running around the clock. I also have an MKS Pi on a polyformer and no issues there either. Basically the cheap wifi dongle they sell with it on AliExpress is unstable. At first I thought it was "a" bad dongle, but I tried all three that came with my Pis and the same results.

onolox, 2023-02-23:
> > > Edit: I purchased a BrosTrend Linux USB WiFi Adapter dongle on Amazon and no more issues
> > 
> > 

What's the chipset of it?

fbeltrao-br, 2023-02-23:
Never buy a wifi dongle before checking if your kernel has the proper drivers. Even if the driver exist in the wild, you won't be able to compile because MKS or whoever built this image did not included the headers, and to make it worst, they used an EOL kernel.

mastercartman, 2023-06-04:
Hey, I bought a tp-link AC1300 Mini Wireless USB Adapter, also named tp-link Archer T3U, for around 18 bucks from Amazon. Just plug and play, works flawless. Got it recommended by Kaiser_E24 in the comments section https://3dpandme.com/2022/09/16/tutorial-mks-pi-klipper-install/comment-page-1/?unapproved=982&moderation-hash=47a08c453741638c05d9707fec751633#comment-982

Waffle0815, 2023-06-17:
I had the same issue with super slow and unstable wifi with the provided wifi adapter.
But I found a super old dongle in my drawer named "Edimax EW-7811UN" which works out of the box without any problems so far. You can get these used for very little money.
Hope this helps.