makerbase-mks/MKS-PI #20: armbianenv.txt missing (causing empty hardware options and UART not working)

Jannomag, 2022-11-07:
Hi,
the armbianenv.txt in /boot/ is missing completely.
This causes that UART didn't work for me until I created it and added this line to the file:
`overlays=uart1`
Since in the readme is written, that UART is also a way to connect the printer to the MKS PI it should be there by default!

Also caused by this file missing is the empty hardware config when running `armbian-config`:
[image]

I don't know if there's any active development since there are old open issues but maybe you'll update this.

Stevexgk9, 2022-12-13:
Could it be the reason why it only recognizes one of my two printers? It seems that both are connected to a single port
/dev/serial/by-id/usb-FTDI_FT232R_USB_UART_AM00N3U0-if00-port0
/dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0

lesbogecko, 2023-02-15:
I added the txt File you suggested but still get nothing connected over UART