makerbase-mks/MKS-PI #22: Tft screen and connection issue

RandomeName, 2022-11-11:
Hi. I just got my mks pi.

The first issue is tft screen won't turn on.
The second issue is. I cant connect mks pi to putty terminal. I follow the instructions. All I got was black terminal and nothing happens when I press enter.

I also tried to clean and flash SD card again and it stil dont work.
And I also tried to check the ribbon connection to the tft screen and it looks good.

georgeion88, 2023-01-25:
I have the same issue , did you find a solution ?

RandomeName, 2023-01-25:
@georgeion88 
Hi .
I have not found a solution for the screen and connection issue.

I have replace the mks pi with Raspberry PI model 4.

georgeion88, 2023-01-25:
Thanks, i think is the best solution and MKS Pi is just waste of money.