makerbase-mks/MKS-PI #23: mks-pi + pi-ts35

psjshep, 2022-11-24:
Using the install system image linked in the instructions ( https://drive.google.com/drive/folders/1tTuSvF9OL2qtPXElau8YOXn2sWbdxa9e?usp=sharing)  & the PI-TS35 touch screen, all installed & I can see Klipper on the screen.

I am unable to get the touchscreen to recognise anything.

I was able to install & test xlibinput_calibrator from https://github.com/fbeltrao-br/mksts35 - this displayed the 4 corners cross & let me touchscreen on them - this proves the hardware works.... if I just used xinput_calibrator, it always complained that the touch was outside the limits... even if I used a stylus accurately!

So is the problem with Klipperscreen or something else?

Phil

J6633257, 2023-03-05:
OMG, i have the same problem, even bought a 2nd one just to make sure I wasn't going crazy, have you found a solution to this? I'm at my wits end

psjshep, 2023-03-05:
Nope... I moved on & used a PI with a bigger touch screen...

Never heard anything from here on github.... looks like they post but don't bother reading the issues....

Good luck... post back if you hear anything

onolox, 2023-03-15:
Try tocontact the [Support](support@makerbase.com.cn).