makerbase-mks/MKS-PI #26: Support MKS PI-TS35 by newest Armbian Kernel

redrathnure, 2022-11-26:
Newest Armbian use 6.x kernel and MKS PI-TS35 does not works anymore. Any plans to update MKS-PI-Armbian image? And it would be nice to publish instructions how to enable MKS PI-TS35 screen with new Armbian (or other distro).

compihu, 2022-12-03:
They will not. How they hacked their SPI TFT to the Linux kernel is a complete mess. They took the [ili9431](https://cdn-shop.adafruit.com/datasheets/ILI9341.pdf) (which is a fixed, 240x320 pixel resolution TFT driver circuit) kernel driver from [fbtft](https://github.com/notro/fbtft), changed the WIDTH and HEIGHT macro according to their resolution (320x480) and released the binary kernel module.
I am in the process of reverse engineering and solving the whole mess correctly but the time I can spend on it is limited.  The ILI9341 driver is still in the staging section of the 6.0 kernel so I don't see any obstacle in making it available in the same (disgusting) way they did, but I yet have to find a correct solution.

majekw, 2022-12-03:
I think you could find MKS patches to Armbian and kernel here: https://github.com/makerbase-mks/armbian-build
LCD changes are in userpatches/kernel/rockchip64-legacy/kernel-rockchip64-legacy.patch

compihu, 2022-12-03:
Thanks, @majekw, this closes the GPL violation bug

redrathnure, 2022-12-07:
Thanks for the info. However, I expect more support from a supplier who makes money from this hardware.

redrathnure, 2022-12-24:
@compihu, @majekw  I tried to apply patches to the current and edge kernels. https://github.com/redrathnure/armbian-mkspi/tree/custom/mkspi_23.02.0-trunk.0073.  The same not really nice way to patch ILI9341 driver, however it works.

redrathnure, 2022-12-26:
oh, this great news! Could you share example of proper device tree declaration?

compihu, 2022-12-27:
@redrathnure, As far as I can see from your commits, you've already figured it out...