makerbase-mks/MKS-PI #28: Can i connect and configure the MKS-PWC V2.0?

Atumra78, 2022-12-06:


mpalpha, 2023-01-08:
how do I connect and configure the mks-pi and mks-pwc with klipper?

Liriel-666, 2023-03-01:
its easy.
Connect it normal

then this in printer.cfg .
You must change the pin because my is for robin nano v3

########################################
# PWC Module
########################################
[output_pin PW_OFF]
# Define output pin to power supply off
pin: PB2
value: 1 # Maintain power supply on

and this too or in a extra macro cfg

#######################################
# PWC Module
########################################

[gcode_macro M80]
description: Shutdown 12V power supply
gcode:
  SET_PIN PIN=PW_OFF VALUE=0

it appears on Dashbard a button PW OFF to manual shut it down.

M80 in the End of the printfile shut down automatic the system

mpalpha, 2023-03-05:
> its easy. Connect it normal
>
> then this in printer.cfg . You must change the pin because my is for robin nano v3
>
> ```
>
> ########################################
> # PWC Module [...]

what about power button on/off detection?

Liriel-666, 2023-03-05:
Button install like normal
Power on over Button works without a Problem.
Power off not but here you can use the button on the web interface or the button on the display for manual shut down

mpalpha, 2023-03-13:
You say "normal" many times.
do you have the "normal" instructions or diagrams?