makerbase-mks/MKS-PI #30: Please provide full kernel patches and Armbian configs (GPL license violation)

redrathnure, 2022-12-09:
[makerbase-mks/armbian-build](https://github.com/makerbase-mks/armbian-build/) repo has 2 commits, but they do not looks like full sources for distributed [Armbian image]( https://drive.google.com/drive/folders/1tTuSvF9OL2qtPXElau8YOXn2sWbdxa9e?usp=sharing).
There are few issues:
1. Seems  [Armbian image]( https://drive.google.com/drive/folders/1tTuSvF9OL2qtPXElau8YOXn2sWbdxa9e?usp=sharing) was built from `edge` version (kernel 5.16.20), but [makerbase-mks/armbian-build](https://github.com/makerbase-mks/armbian-build/) repo has patches only for `legacy` one
2. `config/targets.conf` has declaration of `mkspi` board, but no valid board configs or any related patches were published
3. This still looks like GPL violation, because no sources were open for distributed kernel and  [Armbian image]( https://drive.google.com/drive/folders/1tTuSvF9OL2qtPXElau8YOXn2sWbdxa9e?usp=sharing) (see also [GPL violation: MKS distributes binaries without a valid license #24 ](https://github.com/makerbase-mks/MKS-PI/issues/24) from [compihu](https://github.com/compihu) for more details).

Please publish full sources  (including kernel, u-boot patches and Armbian configs)  and kernel builds, which were used for [Armbian image]( https://drive.google.com/drive/folders/1tTuSvF9OL2qtPXElau8YOXn2sWbdxa9e?usp=sharing)

fbeltrao-br, 2022-12-12:
They even used a EOL kernel, they could at least use a 5.15.y release.

We need the actual kernel and u-boot patches used to compile this image, they are the most critical part, other stuff like kernel config and device trees can be worked around.

redrathnure, 2022-12-24:
@fbeltrao-br [there are attempt to apply patches](https://github.com/redrathnure/armbian-mkspi/tree/custom/mkspi_23.02.0-trunk.0073) to the current Armbian sources. Seems board and TFT work with 5.15 and 6.1 kernels.

fbeltrao-br, 2022-12-25:
@redrathnure yeah, I've successfully ported a few drivers to 5.15.y and had almost everything working, until I found that they used an old SPI driver from kernel 4.4 for Android Rockchip SOC and decided to give up. I'm not going to touch on that.

Touchscreen driver can be ported in kernel without modprobe.

redrathnure, 2022-12-25:
Do you mean `ads7846` one?

Jarewa, 2022-12-25:
> @fbeltrao-br [there are attempt to apply patches](https://github.com/redrathnure/armbian-mkspi/tree/custom/mkspi_23.02.0-trunk.0073) to the current Armbian sources. Seems board and TFT work with 5.15 and 6.1 kernels.

Put the iso image somewhere.

redrathnure, 2022-12-25:
> Put the iso image somewhere.

Any idea where/how put a few files 2-4GB each?

Jarewa, 2022-12-25:
Google, or another server?? 

I'd love to test it..

redrathnure, 2022-12-25:
Release description: https://github.com/redrathnure/armbian-mkspi/releases/tag/mkspi%2F0.1.0_23.02.0-trunk.0073
Google Drive with images: https://drive.google.com/drive/folders/1Qi8wgqquht5gaP7EmcN2nvd1rUnSABRf?usp=sharing

josueturco, 2023-02-23:
@redrathnure how to compile the image? I cloned armbian build files from makerbase but i don't know what hardware to select to compile

redrathnure, 2023-02-23:
@josueturco please read this topic from start. IMO they have not provided any sources which you may use.

If you are interested in a working Armbian, please visit [unofficial fork](https://github.com/redrathnure/armbian-mkspi/releases/tag/mkspi%2F0.1.6_23.02.0-trunk.0236) or [related release page](https://github.com/redrathnure/armbian-mkspi/tree/custom/mkspi_23.02.0-trunk.0236)

fbeltrao-br, 2023-02-26:
Funny how they just ignore us.

All we need are the actual u-boot and kernel user patches, not the fake one they uploaded.

It's literally just two text files.

josueturco, 2023-03-08:
@redrathnure yeah, thanks! I just compile the unofficial fork and this work for me. do you know how to force other resolutions? I am using a HDMI 7" 1024x600 lcd from aliexpress but the screen icons is very small...

josueturco, 2023-03-08:
In the armbian screen config I can see that HDMI is configured at 1920x1080 but my lcd is 1024x600.

No more options on armbian resolutions.

redrathnure, 2023-03-08:
@josueturco, sorry, I have no idea how about HDMI configuration (yet). Perhaps https://docs.armbian.com/Hardware_Allwinner/#how-to-reconfigure-video-output or https://unix.stackexchange.com/questions/408558/how-to-change-display-resolution-for-armbian may give you some hints.

josueturco, 2023-03-10:
@redrathnure OK. I think this is for Allwinner cpu... on the mkspi it's rockchip... I'll keep looking... Thanks!!