makerbase-mks/MKS-PI #32: Rotate PI-TS35 v1.0 90° (vertical)

random924, 2022-12-25:
I have a problem, flying bear ghost 6 has a vertical screen (ts35), i have a mks pi and new (other) ts35 screen. By default, the klipperscreen menu is horizontal. How to rotate menu vertically on mks pi with new ts35 screen?

redrathnure, 2022-12-25:
Most likely you have to edit your device tree. Open rk3328-roc-cc.dts and find following section:
```
		spi_for_lcd@0 {
			compatible = "ilitek,ili9341";
			pinctrl-names = "default";
			pinctrl-0 = <0x32>;
			reg = <0x00>;
			spi-max-frequency = <0x16e3600>;
			bgr;
			fps = <0x14>;
			rotate = <0x5a>;
			buswidth = <0x08>;
			dc-gpios = <0x31 0x06 0x00>;
			reset-gpios = <0x31 0x04 0x01>;
			led-gpios = <0x31 0x05 0x00>;
			status = "okay";
		};
```

Play around `rotate = <0x5a>;` string.

Please refer https://github.com/makerbase-mks/MKS-PI#how-to-modify-the-device-tree-file to get help about `/boot/dtb/rockchip/rk3328-roc-cc.dtb` editing.

random924, 2022-12-25:
Ok, it is works, but touch doesnt work correctly

redrathnure, 2022-12-25:
I would recommend to check https://klipperscreen.readthedocs.io/en/latest/Hardware/#touchscreen-touch-rotation and https://github.com/fbeltrao-br/mksts35 info. They have instruction for touchscreen calibration, especially the first link looks promising.

random924, 2022-12-25:
This works great, thank you very much