makerbase-mks/MKS-PI #33: printer board not visible

Jean-Fred-CEC, 2022-12-31:
Hi,
I am trying to connect my CR10S Pro V1 (flashed with Tiny Machine 1.1.9B7) to the MKS Pi board.
I've flashed my board with "Armbian_22.05.0-linux_5.16.20" and until here everything is OK
Wifi works fine, Fluidd is OK without the printer connection, camera also is working
So I am looking for the printer ID in order to flash the printer firmware with Klipper, but the command "ls /dev/serial/by-id/*"  answers : "No such file or directory"
I've checked my USB ports with lsusb without the printer, and with the printer, so I see clearly the port where it is connected.
I changed the USB cable and there is no change. 
I'm clearly new to Linux and I don't see what's wrong, can someone help me please?

Thanks in advance, best regards,

Jean-Fred

PedroLageTavares, 2023-03-09:
Do you had klipper previously installed or is this the first attempt? 

You can use instead the /dev/ttyUSB0 (can be USB0, USB1 ...) in klipper printer.cfg

also, check this thread: https://klipper.discourse.group/t/e3v2-ls-dev-serial-by-id-not-working/3934

Jean-Fred-CEC, 2023-03-13:
Thanks for replying!
My card fried, I just bought a Raspberry, everything works fine now.

Best regards.

Jean-Fred