makerbase-mks/MKS-PI #34: 5" hdmi touchscreen does not work

TheFeralEngineer, 2023-01-07:
5" hdmi touchscreen (800x480) does not work with this sbc. 7" from same manufacturer (1080x600) works, but if I remove klipperscreen and load the xfce4 desktop, it's forced to 1960x1080 resolution and can't be changed.

redrathnure, 2023-01-08:
Could you try [Custom Build 0.1.4_23.02.0-trunk.0135](https://github.com/redrathnure/armbian-mkspi/releases/tag/mkspi%2F0.1.4_23.02.0-trunk.0135) image and share output of `fbset -i` command?

TheFeralEngineer, 2023-01-08:
> Could you try [Custom Build 0.1.4_23.02.0-trunk.0135](https://github.com/redrathnure/armbian-mkspi/releases/tag/mkspi%2F0.1.4_23.02.0-trunk.0135) image and share output of `fbset -i` command?

sure, here you go. 

mode "480x320"
    geometry 480 320 480 320 16
    timings 0 0 0 0 0 0 0
    nonstd 1
    rgba 5/11,6/5,5/0,0/0
endmode

Frame buffer device information:
    Name        : fb_ili9481
    Address     : 0
    Size        : 307200
    Type        : PACKED PIXELS
    Visual      : TRUECOLOR
    XPanStep    : 0
    YPanStep    : 0
    YWrapStep   : 0
    LineLength  : 960
    Accelerator : No

I don't even think it notices that the 5" is plugged in with this build. On the standard build, it shows the interpolated 1920x1080 resolution, but nothing shows up on the screen. If I have the 7" screen plugged in when I boot up the stock build, it'll show klipperscreen and then if i swap displays, it'll show the upper left corner of klipperscreen, but rebooting results in no picture. I haven't tried that with your custom build yet, but right now, it doesn't really show any sign of life.

TheFeralEngineer, 2023-01-09:
> Could you try [Custom Build 0.1.4_23.02.0-trunk.0135](https://github.com/redrathnure/armbian-mkspi/releases/tag/mkspi%2F0.1.4_23.02.0-trunk.0135) image and share output of `fbset -i` command?

Also, unrelated, but the uart pins won't reset the mcu like the stock firmware will. I have to power off the machine to reset the firmware

redrathnure, 2023-01-09:
> Name : fb_ili9481
> ...
> I don't even think it notices that the 5" is plugged in with this build. 

yeah, looks like MKS TS35 display, but not your 5" one.

> Also, unrelated, but the uart pins won't reset the mcu like the stock firmware will
Have you tried `current_5.15.86` or `edge_6.1.3` image? Reset worked only in `current_5.15.86` version

TheFeralEngineer, 2023-01-09:
> yeah, looks like MKS TS35 display, but not your 5" one.
>
>
> Have you tried `current_5.15.86` or `edge_6.1.3` image? Reset worked only in `current_5.15.86` version

yeah and the TS35 is garbage. I've got one and it barely responds. I wanted to use the 5" because the 7 is huge, but neither build recognizes it. 

I tried the current build and reset didn't work.

Jarewa, 2023-01-14:
/etc/X11/xorg.conf.d/02-driver.conf change to 

Section "Device"
         Identifier "LIMA"
         Driver "modesetting"
EndSection

, if that didn't help, add w /etc/X11/xorg.conf.d/01-armbian-defaults.conf 

Section "Monitor"
        Identifier              "HDMI-A-1"
        Option                  "DPMS" "true"
EndSection
Section "Screen"
        Identifier              "Screen0"
        Device                  "LIMA"
        Monitor                 "HDMI-A-1"
        SubSection              "Display"
               Modes                 "XXXXxXXXX"
        EndSubSection
EndSection

where "XXXXxXXXX"  is the resolution supported by the display, unless EDID works, don't work all in "/var/log/Xorg.0.log" you'll know which ones work because I don't remember anymore.

TheFeralEngineer, 2023-01-14:
> /etc/X11/xorg.conf.d/02-driver.conf change to
>
> Section "Device" Identifier "LIMA" Driver "modesetting" EndSection
>
> , if that didn't help, add w /etc/X11/xorg.conf.d/01-armbian-defaults.conf
>
> Section [...]

Appreciate the help. Looks like those changes my be attempting to do something, but still no joy. Here a clip from the log

[   345.574] 	Module class: X.Org Video Driver
[   345.574] 	ABI class: X.Org Video Driver, version 24.0
[   345.574] (II) modesetting: Driver for Modesetting Kernel Drivers: kms
[   345.576] (II) modeset(0): using drv /dev/dri/card0
[   345.577] (==) modeset(0): Depth 24, (==) framebuffer bpp 32
[   345.577] (==) modeset(0): RGB weight 888
[   345.577] (==) modeset(0): Default visual is TrueColor
[   345.577] (II) Loading sub module "glamoregl"
[   345.577] (II) LoadModule: "glamoregl"
[   345.578] (II) Loading /usr/lib/xorg/modules/libglamoregl.so
[   345.594] (II) Module glamoregl: vendor="X.Org Foundation"
[   345.594] 	compiled for 1.20.4, module version = 1.0.1
[   345.594] 	ABI class: X.Org ANSI C Emulation, version 0.4
[   345.862] (II) modeset(0): Refusing to try glamor on llvmpipe
[   345.865] (EE) modeset(0): glamor initialization failed
[   345.865] (II) modeset(0): ShadowFB: preferred NO, enabled NO
[   345.974] (II) modeset(0): Output HDMI-1 using monitor section HDMI-A-1
[   346.084] (II) modeset(0): EDID for output HDMI-1
[   346.084] (II) modeset(0): Manufacturer: ADA  Model: 4  Serial#: 1
[   346.084] (II) modeset(0): Year: 2007  Week: 1
[   346.084] (II) modeset(0): EDID Version: 1.3
[   346.084] (II) modeset(0): Digital Display Input
[   346.084] (II) modeset(0): Max Image Size [cm]: horiz.: 15  vert.: 10
[   346.084] (II) modeset(0): Gamma: 1.00
[   346.084] (II) modeset(0): No DPMS capabilities specified
[   346.084] (II) modeset(0): Supported color encodings: RGB 4:4:4 YCrCb 4:4:4 
[   346.084] (II) modeset(0): First detailed timing is preferred mode
[   346.084] (II) modeset(0): redX: 0.000 redY: 0.000   greenX: 0.000 greenY: 0.000
[   346.084] (II) modeset(0): blueX: 0.000 blueY: 0.000   whiteX: 0.000 whiteY: 0.000
[   346.084] (II) modeset(0): Manufacturer's mask: 0
[   346.084] (II) modeset(0): Supported detailed timing:
[   346.084] (II) modeset(0): clock: 32.0 MHz   Image Size:  150 x 100 mm
[   346.084] (II) modeset(0): h_active: 800  h_sync: 840  h_sync_end 888 h_blank_end 928 h_border: 0
[   346.084] (II) modeset(0): v_active: 480  v_sync: 493  v_sync_end 496 v_blanking: 525 v_border: 0
[   346.084] (II) modeset(0): Number of EDID sections to follow: 1
[   346.084] (II) modeset(0): EDID (in hex):
[   346.084] (II) modeset(0): 	00ffffffffffff000481040001000000
[   346.084] (II) modeset(0): 	01110103800f0a000a00000000000000
[   346.084] (II) modeset(0): 	00000000000001010101010101010101
[   346.084] (II) modeset(0): 	010101010101800c208030e02d102830
[   346.084] (II) modeset(0): 	d3006c44000000180000001000000000
[   346.084] (II) modeset(0): 	00000000000000000000000000100000
[   346.084] (II) modeset(0): 	00000000000000000000000000000010
[   346.084] (II) modeset(0): 	00000000000000000000000000000116
[   346.084] (II) modeset(0): 	02031471410023098707830100006503
[   346.085] (II) modeset(0): 	0c0010000000001000100a2020202020
[   346.085] (II) modeset(0): 	2020202020200000001000100a202020
[   346.085] (II) modeset(0): 	20202020202020200000001000010120
[   346.085] (II) modeset(0): 	20202020202020202020000000100010
[   346.085] (II) modeset(0): 	0a202020202020202020202000000010
[   346.085] (II) modeset(0): 	00100a20202020202020202020200000
[   346.085] (II) modeset(0): 	000000000000000000000000000000d9
[   346.085] (II) modeset(0): No remaining probed modes for output HDMI-1
[   346.085] (II) modeset(0): Output HDMI-1 connected
[   346.085] (WW) modeset(0): Unable to find initial modes
[   346.085] (EE) modeset(0): Output HDMI-1 enabled but has no modes
[   346.085] (==) modeset(0): Using gamma correction (1.0, 1.0, 1.0)
[   346.085] (EE) modeset(0): No modes.
[   346.085] (II) UnloadModule: "modesetting"
[   346.085] (II) UnloadSubModule: "glamoregl"
[   346.085] (II) Unloading glamoregl
[   346.085] (EE) Screen(s) found, but none have a usable configuration.
[   346.085] (EE) 
Fatal server error:
[   346.085] (EE) no screens found(EE) 
[   346.085] (EE) 
Please consult the The X.Org Foundation support 
	 at http://wiki.x.org
 for help. 
[   346.085] (EE) Please also check the log file at "/var/log/Xorg.0.log" for additional information.
[   346.086] (EE) 
[   346.088] (EE) Server terminated with error (1). Closing log file.

Jarewa, 2023-01-15:
Do you have this display from mks not connected?, give a screenshot as you wrote in these two files.

It helped me with 4.3inch HDMI LCD (B)

TheFeralEngineer, 2023-01-16:
I don't have the MKS display connected. I honestly wish it worked better so I didn't have to do all of this stuff with HDMI screens.

I didn't write anything in, that log was a direct copy and paste from nano. I can get log files from the 7" screen that works and the 5" screen that does not work, if that would be helpful. The screens are from the same manufacturer and I've tested the 5" screen on a windows PC. It works on that.

EmJay276, 2023-06-28:
Did you get the 5" screen working?

TheFeralEngineer, 2023-06-28:
No, I kept getting mcu shutdowns on both of my pi units and switched to
bigtreetech products

Phil T.
The Feral Engineer