makerbase-mks/MKS-PI #36: Failed reset MCU

Jarov8, 2023-01-10:
For three days I have this message: Failed automated reset of MCU 'mcu'
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Error configuring printer
Software installed on a different memory card, the effect is the same.

TheFeralEngineer, 2023-01-11:
did you try to hard reboot your mcu (power off/on)? I had some issues with this on a different armbian build.

onolox, 2023-02-24:
What's the printer? Describe your setup, people don't have crystal ball.
I changed the usb cable.
Next step is to use UART.

Longoria20, 2023-04-30:
I have an ender 3 with the 4.2.2 creality board, I keep getting the same message as well, I have tried multiple cables already. Is there another way to connect my mks to my board?