makerbase-mks/MKS-PI #39: Possible to connect 2 ADXL345 sensoren?

Liriel-666, 2023-01-28:
Possible to connect 2 ADXL345 sensoren?

FabioM3, 2023-03-07:
However, you can also connect two accelerometers simultaneously, though they must be connected to different boards (say, to an RPi and printer MCU board), or to two different physical SPI interfaces on the same board (rarely available). Then they can be configured in the following manner:

[adxl345 hotend]
# Assuming `hotend` chip is connected to an RPi
cs_pin: rpi:None

[adxl345 bed]
# Assuming `bed` chip is connected to a printer MCU board
cs_pin: ...  # Printer board SPI chip select (CS) pin

[resonance_tester]
# Assuming the typical setup of the bed slinger printer
accel_chip_x: adxl345 hotend
accel_chip_y: adxl345 bed
probe_points: ...

Then the commands TEST_RESONANCES AXIS=X and TEST_RESONANCES AXIS=Y will use the correct accelerometer for each axis.

FabioM3, 2023-03-07:
However, you can also connect two accelerometers simultaneously, though they must be connected to different boards (say, to an RPi and printer MCU board), or to two different physical SPI interfaces on the same board (rarely available). Then they can be configured in the following manner:

```
[adxl345 hotend]
# Assuming `hotend` chip is connected to an RPi
cs_pin: rpi:None

[adxl345 bed]
# Assuming `bed` chip is connected to a printer MCU board
cs_pin: ...  # Printer board SPI chip select (CS) pin

[resonance_tester]
# Assuming the typical setup of the bed slinger printer
accel_chip_x: adxl345 hotend
accel_chip_y: adxl345 bed
probe_points: ...

```
Then the commands TEST_RESONANCES AXIS=X and TEST_RESONANCES AXIS=Y will use the correct accelerometer for each axis.

ColdBeer72, 2023-03-07:
Hey @FabioM3 ! Nice solution, the only trouble is to know how to connect the adxl345 sensor to a mcu board (specially thinking on MKS boards). Do you know how to do it?

onolox, 2023-03-07:
This is the doc of klipper.
In a summary, you need 2 SPI interfaces.

fbeltrao-br, 2023-03-21:
Buy a RP2040 based board and use it as a secondary MCU for accelerometers, it has two SPI interfaces, I've done that when I used to have a cartesian printer