makerbase-mks/MKS-PI #40: Need Help with UART Connection

lesbogecko, 2023-02-15:
Hello,

I really struggling to Connect my MKS Pi with an skr Pico over UART.
Tested with a Pi3A so it is Not caused by the Pico Motherboard.

Can someone please Provide the solution

redrathnure, 2023-02-15:
https://github.com/redrathnure/armbian-mkspi/releases :) These images were tested with  SKR MINI E3 V2.0. via UART interface.

lesbogecko, 2023-02-16:
i already tried them without being successful. is there anything i have toset to make it work? since it says "UART0 not tested yet" i concluded there could be a armbian related issue rather than a mkspi related. i also created a armbianEnv.txt to enable UART1 overlay but i still get a MCU error in serial klippy python scripts

redrathnure, 2023-02-16:
@lesbogecko I would recommend to double check your cable, especially RX/TX pins connection. And pay attention that it's UART**0** which is enabled by default.
I did foloowing steps for MKS mini E3
1. Build and flash Klipper firmware. Just in case I explicitly specified baudrate (250000).
2. Connect MKSPI <--> printer board using 3 wires cable (GND <--> GND, **RX**<-->**TX**, TX<-->RX)
3. Power up all this staff and try to configure Klipper using:
```
[mcu]
serial:/dev/ttyS0
baud: 250000
```
I am explicitly specified baudrate during klipper firmware build and in cfg file, perhaps it isn't necessary.

After klipper reboot it manage to see printer board. 
If you changed DST or armbianEnv.txt I would recommend to start from the scratch (flash new image and repeat configuration steps).

lesbogecko, 2023-02-16:
Thanks a Lot thats the Kind of answer i was looking for, by someone who did it himself and with minor diffs in what i did.Will try it immediatly and give an update.

lesbogecko, 2023-02-16:
i tried like you told still ending up with that MCU error message in log:;

mcu 'mcu': Starting serial connect
mcu 'mcu': Timeout on connect
mcu 'mcu': Wait for identify_response
Traceback (most recent call last):
  File "/home/mks/klipper/klippy/serialhdl.py", line 68, in _get_identify_data
    params = self.send_with_response(msg, 'identify_response')
  File "/home/mks/klipper/klippy/serialhdl.py", line 261, in send_with_response
    return src.get_response([cmd], self.default_cmd_queue)
  File "/home/mks/klipper/klippy/serialhdl.py", line 318, in get_response
    self.serial.raw_send_wait_ack(cmds[-1], minclock, reqclock,
  File "/home/mks/klipper/klippy/serialhdl.py", line 253, in raw_send_wait_ack
    self._error("Serial connection closed")
  File "/home/mks/klipper/klippy/serialhdl.py", line 61, in _error
    raise error(self.warn_prefix + (msg % params))
serialhdl.error: mcu 'mcu': Serial connection closed

i also switched the uart wires in case they are crossed but since then it just says klippy not connected instead of MCU error

redrathnure, 2023-02-16:
I may check my boar with bullseye edge, most likely tomorrow. However either it's non match boudrate or some problem wit MKS PI itself.

BTW, MKS manual [has a chapter for UART connection](https://github.com/makerbase-mks/MKS-PI#uart-connection), so even MKS image should work. If not , perhaps it's another one sign of failed plate.

lesbogecko, 2023-02-16:
the problem i had with the mks image is that after updating klipper components i get a internal host error on fluidd and cant edit or upload config files as well as the configuration section gets empty, no matter whether i update through fluidd, klipperscreen or kiauh, but will try it with the as-it-is mks image just to track down the cause.

thanks in advance i really appreciate your help

lesbogecko, 2023-02-17:
i am not a programmer myself but if i read the log correctly the klipper firmware mcu config has spi0a given gpio0 pin and "CONFIG_HAVE_GPIO_SPI=y". doesnt it mean that it would block the pin to be usable with uart0?

redrathnure, 2023-02-17:
I checked Debian image with SKR board over UART0 connection. Everything works fine. Most likely you should double check Klipper firmware settings. 

> i am not a programmer myself but if i read the log correctly the klipper firmware mcu config has spi0a given gpio0 pin and "CONFIG_HAVE_GPIO_SPI=y". doesnt it mean that it would block the pin to be usable with uart0?

I have never tried to edit these files manually. kiauh should be enough to configure, build and upload MCU firmware.

lesbogecko, 2023-02-17:
my log error looks like this:

Build file /home/mks/klipper/klippy/../out/klipper.dict(8187): Fri Feb 17 05:53:34 2023
Last MCU build version: v0.10.0-530-g3387a9c2-dirty-20230217_055334-mkspi
Last MCU build tools: gcc: (15:7-2018-q2-6) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907] binutils: (2.31.1-12+11) 2.31.1
Last MCU build config: BUS_PINS_spi0a=gpio0,gpio3,gpio2 BUS_PINS_i2c0h=gpio28,gpio29 BUS_PINS_spi0c=gpio16,gpio19,gpio18 BUS_PINS_spi0d=gpio20,gpio23,gpio22 BUS_PINS_i2c0b=gpio4,gpio5 BUS_PINS_i2c0c=gpio8,gpio9 BUS_PINS_i2c0a=gpio0,gpio1 BUS_PINS_i2c0f=gpio20,gpio21 BUS_PINS_i2c0g=gpio24,gpio25 BUS_PINS_i2c0d=gpio12,gpio13 BUS_PINS_i2c0e=gpio16,gpio17 CLOCK_FREQ=12000000 STEPPER_BOTH_EDGE=1 SERIAL_BAUD=250000 ADC_MAX=4095 PWM_MAX=255 BUS_PINS_spi0b=gpio4,gpio7,gpio6 BUS_PINS_i2c1a=gpio2,gpio3 BUS_PINS_i2c1c=gpio10,gpio11 BUS_PINS_i2c1b=gpio6,gpio7 BUS_PINS_i2c1e=gpio18,gpio19 BUS_PINS_i2c1d=gpio14,gpio15 BUS_PINS_i2c1g=gpio26,gpio27 BUS_PINS_i2c1f=gpio22,gpio23 BUS_PINS_spi1c=gpio24,gpio27,gpio26 BUS_PINS_spi1b=gpio12,gpio15,gpio14 BUS_PINS_spi1a=gpio8,gpio11,gpio10 STATS_SUMSQ_BASE=256 RECEIVE_WINDOW=192 MCU=rp2040
Build file /home/mks/klipper/klippy/../out/klipper.elf(355376): Fri Feb 17 05:53:43 2023
mcu 'mcu': Wait for identify_response
Traceback (most recent call last):
  File "/home/mks/klipper/klippy/serialhdl.py", line 69, in _get_identify_data
    params = self.send_with_response(msg, 'identify_response')
  File "/home/mks/klipper/klippy/serialhdl.py", line 258, in send_with_response
    return src.get_response([cmd], self.default_cmd_queue)
  File "/home/mks/klipper/klippy/serialhdl.py", line 316, in get_response
    cmd_queue)
  File "/home/mks/klipper/klippy/serialhdl.py", line 250, in raw_send_wait_ack
    self._error("Serial connection closed")
  File "/home/mks/klipper/klippy/serialhdl.py", line 62, in _error
    raise error(self.warn_prefix + (msg % params))
error: mcu 'mcu': Serial connection closed

also the skr pico is based on rp2040 which the mini isnt so i guess there might be the cause smh

lesbogecko, 2023-02-18:
i am now going to to download the klipper folder from my working Pi3A+ OCTOKLIPPER installation and take a look at the diffs.
must be a difference in between raspbian and armbian for sure...

lesbogecko, 2023-02-18:
i tracked down the issue at least to the point knowing its  neither related to mks pi nor armbian.

made a clean install on my pi3a+ and isntalled everythin over kiauh and tadaa... does not work. so this issue seems to be somewhat with rp2040 and klipper...

so since its the wrong place i should close it i guess. thanks for your help anyway, helped with the conclusion

closed 2023-02-18 by lesbogecko