makerbase-mks/MKS-PI #46: Ender 3 S1 to UART connection does not (seem to) work

remonvv, 2023-03-13:
I'm sure I'm doing something wrong but is there a guide available somewhere? I have the relevant J711 pins from the Ender mobo routed to the G/TX/RX pins but I think I have the wrong configuration somewhere. The guide in the README.MD does not help (just changing the [mcu] to /dev/serial/tty0 doesn't do much)

TheFeralEngineer, 2023-03-15:
first thing's first, you need to compile a proper binary file to use the UART connection on your motherboard through make menuconfig. It's usually best to find the printer cfg file in the klipper config folder (if there is one) and follow the instructions on how to create the file (pins, baud, bootloader offset, etc). Remove the old screen from the mcu before flashing because that can cause an issue and make sure your printer.cfg is proper before turning the machine on because if it connects, you might go into thermal runaway. 

Also, make your MCU section look like this. UART caps at 115200, so if you have random dropouts, you'll have to compile a bin with that rate and force feed it through your printer.cfg:
[mcu]
serial: /dev/ttyS0
restart_method: command
baud: 115200

redrathnure, 2023-04-14:
FYI looks like UART for this board is broken in the latest kernels (both 5.x and 6.x). So I would advice 1. try old image and 2. try non official image where UART functionality is 100% tested.

TheFeralEngineer, 2023-04-14:
I haven't run into uart issues with the latest version