makerbase-mks/MKS-PI #48: MKS PI-TS35 white screen with 20230301 image

fbeltrao-br, 2023-03-20:
Honest question: do you guys even test the images after compiling?

**MKS-PI-Armbian_22.05.0-linux_5.16.20-20220726.img** works with **MKS PI-TS35 V1.0** screen, but the "newer" compiled image **Armbian-makerbase-k5.16.20-EN-20230301.img** all I get is a white screen after boot.

Whats the problem?

fbeltrao-br, 2023-03-21:
Turns out it was caused by the screen + accelerometer. 

You can't use both MKS PI-TS35 and ADXL345 at same time.

If the screen is connected, when you connect ADXL345 SDA or SCL lines, the MKS Pi freezes and the screen turns blank.

If you need one accelerometer, use an Arduino Nano as secondary MCU, or a RP2040 board for two.

I wonder if this is a hardware or software issue. If it's caused by bad software, we can only hope this will be fixed. Judging by how Makerbase don't care much about this, I guess we are hopeless.

DaTTcz, 2023-04-18:
For me, I just need to connect the cable, I don't even need to have the ADXL345 connected to the end of the cable. And the screen of the MKS PI-TS35 is white.

closed 2023-04-28 by fbeltrao-br