makerbase-mks/MKS-PI #50: possible use with ender 3v2 ?

duongnv0499, 2023-04-21:
Hi team, thanks for the great work. I have an Ender 3v2 and wondering that is there a way to use MKS-PI with Ender 3v2 to run Klipper. If it possible, is there any document to do that? Thank you so much !

fbeltrao-br, 2023-04-24:
MKS Pi can run any 3d printer which board can have Klipper flashed in its MCU.

But I'd recommend you to use another SBC with more mature Linux system, like Orange Pi Zero 2.

See the other threads and our struggle to get MKS Pi work properly with touchscreen, input shapping and wi-fi.

virpz, 2023-04-27:
I have it running on a E3 v2 with sprite extruder and adxl345 plus USB webcam. Sure, the MKS pi could use a more recent and refined OS build but then we have [these guys](https://github.com/redrathnure/armbian-mkspi) doing exactly what MKS should have done already . I have no problems with input shaping and In fact I can do 170mm/s at 5k accel and get better results than anything 50mm/s I have ever achieved  with marlin. If you need a config file or some help just ask.

duongnv0499, 2023-04-28:
> I have it running on a E3 v2 with sprite extruder and adxl345 plus USB webcam. Sure, the MKS pi could use a more recent and refined OS build but then we have [these guys](https://github.com/redrathnure/armbian-mkspi) doing exactly what MKS should have done already . I have no problems with input shaping and In fact I can do 170mm/s at 5k accel and get better results than anything 50mm/s I have ever achieved with marlin. If you need a config file or some help just ask.
> 
> The best document is klipper install and config guide https://www.klipper3d.org/Installation.html

Thank you so much ! can I have your email for easier to ask something?

closed 2023-04-28 by duongnv0499