makerbase-mks/MKS-PI #52: Please provide image based on Bullseye release (Debian 11) instead of Buster (Debian 10)

KittenRaee, 2023-05-24:
Since Debian Buster (Debian 10) has reached End of Life and End of Support, it should no longer be used. As new updated versions of Klipper interfaces Mainsail and Fluidd are phasing out support for Buster and require Bullseye to be in place, providing a Buster based image is problematic and requires large amount of work from an end user to keep system up do date and running, especially where amateur Linux users can make a lot of mistakes.
For example Crowsnest >4.0.0 fully requires Bullseye and camera will no longer work if kept up to date.

BenkhoeKK, 2023-06-01:
use firmware Armbian-makerbase-k5.16.20-EN-20230321.img this without issue .

fryc88, 2023-06-21:
But image Armbian-makerbase-k5.16.20-EN-20230321.img is still based on buster, so that does not answer his concern...

jcheger, 2023-06-24:
BTT is much more serious, since they appear in Armbian. I do regret boughting MKS, with such a lack of support. With no response from MKS during one month on this issue, I understand they don't care maintaining this product.
Hopefully, it seams there is an alternative here: https://github.com/redrathnure/armbian-mkspi

redrathnure, 2023-06-24:
FYI please check [release page](https://github.com/redrathnure/armbian-mkspi/releases/tag/mkspi%2F0.1.11_23.05.0-trunk), perhaps you will find [something useful](https://github.com/redrathnure/armbian-mkspi/releases/download/mkspi%2F0.1.11_23.05.0-trunk/Armbian_23.05.0-trunk_Mkspi_bullseye_edge_6.1.28.img.xz) here.

VORONENTHUSIAST, 2023-07-18:
I'm in the middle of upgrading manually using instructions from this link https://www.cyberciti.biz/faq/update-upgrade-debian-10-to-debian-11-bullseye/

I will let you know how it goes!