makerbase-mks/MKS-PI #53: BIGTREETECH ADXL 345 Accelerometer

M1k1m40, 2023-06-09:
I´m trying connect a BIGTREETECH ADXL 345 Accelerometer.
[image]
[image]
I have connected it as follows:
[image]
When I send ACCELEROMETER_QUERY Klipper answer:
// Invalid adxl345 id (got ff vs e5).
// This is generally indicative of connection problems
// (e.g. faulty wiring) or a faulty adxl345 chip.
Any suggestions?

DavidKrall88, 2023-06-15:
Why ADXL345 5V pin connected to MKS PI 3.3V pin. I think the voltage is too low for ADXL... Try to use 5V, what required for that module...

M1k1m40, 2023-06-16:
Yes, I try connect in 5V pin and same issue.