makerbase-mks/MKS-PI #56: I am having a problem connecting MKS pi in my Artillery Sidewinder x2

lone182, 2023-06-16:
Hello everyone, I am having trouble to flash klipper in my 3dprinter. "dfu not detaching" error message. Anyone knows how to fix it?

Jannomag, 2023-07-17:
I had a similar issue, if I remember it correctly. Try to flash [this ](https://www.thingiverse.com/thing:5186707)firmware first, this will enable the M997 command.

ciroanacleto, 2023-07-24:
> Hello everyone, I am having trouble to flash klipper in my 3dprinter. "dfu not detaching" error message. Anyone knows how to fix it?

I had similar problem after configure my sidewinder x2 in a DFU mode and tried to flash the new firmware using STM32 programmer on liux pc. The computer simply did not recognize the printer as connected in USB. The problem was a permission for my user that was running STM32 programmer. I ran as root and it worked.
I don't know if it is the same problem you are facing, but it worth the try.