makerbase-mks/MKS-PI #58: mks pinout

FPPA3D, 2023-06-26:
Hello i just installed MKS PI and most is working with an skr 1.4, until i tried to install the SKR PWC 3.0 on it. i con not find anywere the pinout to use. on raspberry my config was. 

[power Printer]
type: gpio
pin: gpio14
on_when_job_queued: True
off_when_shutdown: True
locked_while_printing: True
restart_klipper_when_powered: True
initial_state: off

and on mooraker

[gcode_macro M81]
gcode:
  {action_call_remote_method("set_device_power", device="Printer", state="off")}

and worked fine. but i tried the pinout for MKS PI and cont get it to work.

i tried the GPIO on tp in this image but did not get it to work. 

Any ideas??
[image]

Thanks

Luigi084, 2023-06-28:
Hello !
I have a problem like that, I use my MKS Pi with a printer that doesn't have a Filament runout sensor. I would like to Wire it with GND / +5V / and 1-C7 pin but I can't find a way to get it's pin number in klipper.

I know that it should work but, we'll need the klipper pin numbers of the reserved IO.

If someone have this, it could be great !

fbeltrao-br, 2023-08-09:
> Hello ! I have a problem like that, I use my MKS Pi with a printer that doesn't have a Filament runout sensor. I would like to Wire it with GND / +5V / and 1-C7 pin but I can't find a way to get [...]

1-C7 is GPIO23.

What board you are using? You can literally use any free header with control pin and GND, such as the ones for endstops. Just because a header is not labeled as filament runout sensor, it doesn't mean it cannot be used for this purpose.

4uva4ek, 2023-09-26:
> 1-C7 is GPIO23.
>
> What board you are using? You can literally use any free header with control pin and GND, such as the ones for endstops. Just because a header is not labeled as filament runout [...]

Where i can get fully information about pinout?