makerbase-mks/MKS-PI #64: Can not install crowsnest because Debian 10.

acguardia, 2023-09-20:
Debian 10 (buster) has reached end of life.

Therefore, crowsnest has also ended support for it.

Please upgrade to a supported OS release,
we are sorry for the inconvenience.

jcheger, 2023-09-22:
duplicate of #52
There is a working alternative image on: https://github.com/redrathnure/armbian-mkspi. But be careful, if the kernel is upgraded, the ethernet interface and the screen won't work anymore (appeared on the current week). See https://github.com/redrathnure/armbian-mkspi#package-updates-via-apt-update how to freeze the kernel updates.

bocoroth, 2024-06-21:
If anyone else is looking for a simpler workaround which doesn't involve upgrading debian, I was able to manually force KIAUH to install the legacy/v3 branch of crowsnest. It is of course outdated but at least it works!

1. Delete the default version of KIAUH or move it to a backup: `cd ~ && mv kiauh/ kiauh_old/`
2. Clone the official KIAUH repo: `git clone https://github.com/dw-0/kiauh.git`
3. `nano kiauh/scripts/crowsnest.sh` and change the branch name on line 37 from `master` to `legacy/v3`. Save and close.
4. Delete any existing crowsnest repo clones: `sudo rm -r crowsnest/`
5. Run `kiauh/kiauh.sh` and install crowsnest as usual.