makerbase-mks/MKS-PI #68: Incorrect System Time

QuinnDamerell, 2024-01-24:
Hello! I'm Quinn, the developer behind OctoEverywhere.com. 

I have had quite a lot of support tickets come in recently with users trying to install OctoEverywhere on MKS PI, but the install fails because the clock is so far out of sync that the SSL handshakes fail.

I'm sure some devices will be out of sync because they might never connect to the internet. But would it be possible to do an NTP time sync for devices that connect to the internet? I'm sure the wrong-time SSL error would also affect other things a user would try to do, so it would be great to fix it if possible!

ColdBeer72, 2024-01-24:
Hi, @QuinnDamere ll!!! I wish you the best but, sincerely, MKS abandoned client support on GitHub months ago.

QuinnDamerell, 2024-01-27:
Oh man, well thanks for letting me know @ColdBeer72! Oddly, the issue seems to be even worse than an out-of-sync clock; the clock seems to always re-sync itself to the 20-day old incorrect date! 

I guess I will look for a word around. :)

fbeltrao-br, 2024-07-20:
There is no RTC on MKS Pi, so time has to sync every time system boots.

Use this:
```
timedatectl list-timezones | grep America
timedatectl set-timezone your_time_zone
timedatectl set-ntp true
```