makerbase-mks/MKS-PI #72: Python is not updating on the latest version

leleomotta, 2024-06-04:
When updating to the latest version of KlipperScreen, the Python version gives an error. The installed version is 3.7.3 and the minimum requirement is 3.8.
There is no point running apt upgrade and update as the version is not updating correctly.
And it's still causing problems in the buster-backports repository (deb http://deb.debian.org/debian buster-backports main contrib non-free)

tup8806, 2024-06-13:
Did you find a fix for this because I'm having the same problem

IgorVoznyj, 2024-06-14:
Did you find a fix for this because I'm having the same problem

leleomotta, 2024-06-21:
I didn't find the ideal solution.
What I did was reinstall everything and didn't update KlipperScreen.

lokertr, 2024-06-23:
Not a fix, but a work around to keep KlipperScreen functional.  Go ahead and edit screen.py in /home/mks/KlipperScreen
on line 147 change min_ver = (3, 8) to min_ver = (3, 7). 

Then open a terminal window and enter 
cd KlipperScreen
then
git reset --hard v0.4.1

KlipperScreen should come back to life.

leleomotta, 2024-06-25:
Hello.
Thanks a lot for the help.
It really worked and now you can use the latest version of KlipperScreen!

nervino666, 2024-06-25:
> Not a fix, but a work around to keep KlipperScreen functional. Go ahead and edit screen.py in /home/mks/KlipperScreen on line 147 change min_ver = (3, 8) to min_ver = (3, 7).
>
> Then open a terminal [...]

I followed your advice, but I can't install KlipperSreen again, it gives me a Python 3.8 nn found error.

nervino666, 2024-06-25:
> Not a fix, but a work around to keep KlipperScreen functional. Go ahead and edit screen.py in /home/mks/KlipperScreen on line 147 change min_ver = (3, 8) to min_ver = (3, 7).
>
> Then open a terminal [...]

Hi, I followed your advice, but I can't install KlipperSreen again, it gives me a Python 3.8 nn found error.

fbeltrao-br, 2024-07-20:
`sudo add-apt-repository ppa:deadsnakes/ppa
sudo apt update
sudo apt install python3.10 python3.10-dev python3.10-venv`

leleomotta, 2024-08-05:
After running the command "sudo add-apt-repository ppa:deadsnakes/ppa" the update gives an error:

Hit:1 http://archive.debian.org/debian buster-backports InRelease
Hit:2 http://deb.debian.org/debian buster InRelease
Hit:3 http://deb.debian.org/debian buster-updates InRelease
Hit:4 http://security.debian.org buster/updates InRelease
Ign:5 http://ppa.launchpad.net/deadsnakes/ppa/ubuntu oracular InRelease
Err:6 http://ppa.launchpad.net/deadsnakes/ppa/ubuntu oracular Release
  404  Not Found [IP: 2620:2d:4000:1::81 80]
Reading package lists... Done
E: The repository 'http://ppa.launchpad.net/deadsnakes/ppa/ubuntu oracular Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.

nervino666, 2024-08-05:
Grazie ho risolto installando armbian per mkspi, ora va benissimo.

swalsh11, 2024-11-18:
you need a new install that supports python 3.8 or better.
Get a new class A2 ssd card.
backup your print.cfg files and any other customised configs.
Download bullseye or bookworm image from: https://github.com/redrathnure/armbian-mkspi/releases. 
Flash new ssd ssd with Blaena Etecher 
Edit d:\dtb\rockchip/rk3328-roc-cc.dtb to fix lcd rotation.  
on First boot set default user name mks.
Full install instructions: https://github.com/redrathnure/armbian-mkspi

plug in network cable and ssh into the machine with putty

Install kiauh by entering this command.
cd ~
git clone https://github.com/th33xitus/kiauh.git
./kiauh/kiauh.sh

in Kiauh install: 
Klipper, moonraker, mainsail/ fluid, klipperscreen,  shell command, & klipper backup. 
flash mcu with linux process   https://www.klipper3d.org/RPi_microcontroller.html
Copy over you old print.cfg files to new machine