makerbase-mks/MKS-PI #73: interface on mks pi

Toxin120, 2025-01-09:
Tell me, can I use both can and serial interface on mks pi?

redrathnure, 2025-01-09:
If you are talking about PI <--> MCU connection, most likely no. You should cheese only one of options during Klipper compilation (usb, uart or usb-can bridge).
However you are free to use unused UART ports to connect other devices.

Toxin120, 2025-01-09:
Can I continue in Russian?

Toxin120, 2025-01-09:
the task is to connect the mcu via the serial interface and the print head via CAN.

redrathnure, 2025-01-09:
I prefer EN because many peoples use this info for troubleshooting and ru-en translation is extremely crappy :)

So, if you looking for the following topology, then no, unfortunately:
```
PI <- UART -> MCU <- CAN -> HT
```
Klipper does not support UART to CAN bridge mode. Feel free to check Klipper forum for more details (e.g. [this one](https://klipper.discourse.group/t/feature-request-uart-to-can-bridge-mode-for-stmgob1-boards/8776/5))

However you may implement it with additional external USB to CAN board:
```
PI <-USB-> USB_TO_CAN_adapter <-CAN-> TH
PI <-UART-> MCU
```

Toxin120, 2025-01-09:
Current topology (by equipment models): MKS Robin nano v3.1<-serial->MKS PI v 1.1<-usb ->MKS UTC v 1.0<-CAN ->MKS THR 42 rev.1.0. I can't figure out how to make CAN work.

redrathnure, 2025-01-09:
> MKS Robin nano v3.1<-serial->MKS PI v 1.1

## Robin Nano part:
```
mkdir ~/firmware

cd ~/klipper/
make menuconfig
# Se settings bellow
make clean && make

# Rename  ~/klipper/out/klipper.bin to  Robin_nano_v3.bin
# Copy Robin_nano_v3.bin klipper firmware to TF card and insert board
#  Reset or repower the board and update firmware
```
[setting for Klipper from MKS repo ](https://github.com/makerbase-mks/Klipper-for-MKS-Boards/blob/main/MKS%20Robin%20Nano%20V3.x/Image/MKS%20Robin%20Nano%20V3.x%20setting%20on%20USART3.png)

## MKS PI part:

Fragment from printer.cfg:
```
[mcu]
serial:/dev/ttyS0
baud: 250000 
```
Boards should be connected:
MKSPI: UART0 (pins next to HDMI slot)
RN3: J2/UART3 (pins next to wifi module)

## MKS PI v 1.1<-usb ->MKS UTC v 1.0<-CAN ->MKS THR 

You need to flash MKS UTC v 1.0 with needed baud rate. I prefer 1M, he same for MKS THR. 

MKSPI part: may get some info from [from readme](https://github.com/redrathnure/armbian-mkspi?tab=readme-ov-file#how-to-configure-can-bus)
MKS UTC v 1.0 and MKS THR: please google for Klipper firmware configuration or use from MKS [repos ](https://github.com/makerbase-mks/MKS-THR36-THR42-UTC/tree/main/THR%20firmware(Can)) (e.g.  and ).

Toxin120, 2025-01-09:
Thank you, your information helped to solve the problem

closed 2025-01-09 by Toxin120

redrathnure, 2025-01-09:
@Toxin120 BTW you may try https://github.com/redrathnure/armbian-mkspi/releases or official Armbian images. They are more updated, however you would need to install Klipper from scratch e.g. via [kiauh](https://github.com/dw-0/kiauh).